ARM-software/CMSIS-FreeRTOS #3: cmsis_os2 header

euripedesrocha, 2017-04-21:
Is cmsis_os2.h available? 

I'm trying to test CMSIS Freertos but the file is missing. I searched in the release and repository.

jkrech, 2017-04-24:
The file cmsis_os2.h contains the CMSIS-RTOS2 API. The specification of this RTOS and it's API is part of the ARM.CMSIS pack. 
https://github.com/ARM-software/CMSIS_5
folder:
https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/RTOS2/Include

Why is the include file not in this repository ?
CMSIS-Pack (documented here http://arm-software.github.io/CMSIS_5/Pack/html/index.html) has the benefit that software is desribed in APIs and Software components.

The CMSIS-Pack description file:
ARM.CMSIS-FreeRTOS.pdsc
specifies a dependency on a minimum version of the CMSIS pack:
<requirements>
    <packages>
      <package vendor="ARM" name="CMSIS" version="5.0.1-0"/>
    </packages>
</requirements>

it furthermore specifies a component:
<component Cclass="CMSIS" Cgroup="RTOS2" Csub="FreeRTOS" Cversion="9.0.0" Capiversion="2.1.0" condition="FreeRTOS RTOS2 Core">
implementing version 2.1.0 of the CMSIS:RTOS2 API, which is defined in the above referenced ARM.CMSIS pack version 5.0.1-0 or higher.

The Runtime Environment automatically setup the compiler include path to include cmsis_os2.h.

JonatanAntoni, 2017-08-11:
I guess this question was answered completely thus I close it now. If you have additional remarks please feel free to reopen it and continue posting here.

closed 2017-08-11 by JonatanAntoni