ARM-software/CMSIS-FreeRTOS #4: osThreadDef macro invalid reference

aport, 2017-06-02:
Trying to use the `osThreadDef` results in the following error:
```main.c:7:13: error: 'blinker' undeclared here (not in a function)
 osThreadDef(blinker, osPriorityNormal, 1, 256);
             ^
../../cmsis_os.h:438:4: note: in definition of macro 'osThreadDef'
 { (name), \
    ^
```

The first field in the `osThreadDef_t` structure is a function pointer, which is unused in the FreeRTOS implementation. In addition, the macro sets the thread name to NULL.

Modifying the definition of `osThreadDef` to the following:

```
#define osThreadDef(name, priority, instances, stacksz) \
static uint64_t os_thread_stack##name[(stacksz)?(((stacksz+7)/8)):1]; \
static StaticTask_t os_thread_cb_##name; \
const osThreadDef_t os_thread_def_##name = \
{ NULL, \
  { #name, osThreadDetached, \
    (instances == 1) ? (&os_thread_cb_##name) : NULL,\
    (instances == 1) ? sizeof(StaticTask_t) : 0U, \
    ((stacksz) && (instances == 1)) ? (&os_thread_stack##name) : NULL, \
    8*((stacksz+7)/8), \
    (priority), 0U, 0U } }
#endif
```
Resolves the issue.

RobertRostohar, 2017-06-07:
Existing osThreadDef macro should work.

> The first field in the  osThreadDef_t structure is a function pointer, which is unused in the FreeRTOS implementation.

Not really. Function pointer is required and defines the thread execution starting point.

Have you defined your thread function?

Example:

```
void blinker (void const *arg) {
  // ...
}

osThreadDef(blinker, osPriorityNormal, 1, 256);
```

Anyway `osThreadDef `is only used for the legacy RTOS1 API. RTOS2 API should be used for new projects.

aport, 2017-06-07:
You are correct, thank you

closed 2017-06-07 by aport