ARM-software/CMSIS-FreeRTOS #7: No MemPool functions

InhyungLee, 2017-07-27:
Why did you not implement some function for cmsis_os2.h?
For instance, mempool, thread, kernel... like followings..

Memory Pool functions are not implemented and will always return with error.
   - \b osMemoryPoolAlloc: \token{not implemented}
   - \b osMemoryPoolDelete: \token{not implemented}
   - \b osMemoryPoolFree: \token{not implemented}
   - \b osMemoryPoolGetBlockSize: \token{not implemented}
   - \b osMemoryPoolGetCapacity: \token{not implemented}
   - \b osMemoryPoolGetCount: \token{not implemented}
   - \b osMemoryPoolGetName: \token{not implemented}
   - \b osMemoryPoolGetSpace: \token{not implemented}
   - \b osMemoryPoolNew: \token{not implemented}

JonatanAntoni, 2017-08-11:
Hi @InhyungLee,

There are some functionality that is now supported natively by FreeRTOS and we decided to not implement stuff externally. 

Regarding memory pools in FreeRTOS there is a statement from Richard on [stackoverflow](https://stackoverflow.com/questions/37547028/memory-pool-in-freertos-like-in-uc-os-ii).

Do you have dedicated requirements you would like to see addressed by CMSIS-FreeRTOS?

Regards,
Jonatan

InhyungLee, 2017-10-10:
Hello @JonatanAntoni 

Thank you for your response.
I respect your decision. 

Regards,
Inhyung

closed 2017-10-10 by InhyungLee