ARM-software/CMSIS-FreeRTOS #10: osDelayUntil implementation is wrong

CppBaddy, 2018-03-08:
According to RTOS2 API ticks parameter specifies absolute time in ticks to wait for. See my comments in the code excerpt:
osStatus_t osDelayUntil (uint32_t ticks) {
  TickType_t tcnt;
  osStatus_t stat;

  if (IS_IRQ()) {
    stat = osErrorISR;
  }
  else {
    stat = osOK;
    tcnt = xTaskGetTickCount();

    //vTaskDelayUntil (&tcnt, (TickType_t)ticks); <- second parameter must be ticks to wait
    vTaskDelayUntil (&tcnt, (TickType_t)(ticks - tcnt));  // proposed fix
  }

  return (stat);
}

JonatanAntoni, 2018-03-09:
Hi @CppBaddy,

thanks for making us aware of this issue. 

Cheers,
Jonatan

ReinhardKeil, 2018-05-28:
Issue is fixed. Closing as there is no further activity.

closed 2018-05-28 by ReinhardKeil