ARM-software/CMSIS-FreeRTOS #11: cmsis-os2 priorities issue with the FreeRTOS

rahmanih, 2018-03-27:
Hi,
the cmsis-os2.h defines priorities going from:
`osPriorityNone          =  0,         ///< No priority (not initialized).`
to
` osPriorityISR           = 56,         ///< Reserved for ISR deferred thread.`

FreeRTOS, however, is customising the priorities using the `configMAX_PRIORITIES` that **can't** be set 
higher than 32, otherwise a compile error occurs, in `portmarco.h` files.
 Added to that, when a thread is created with a priority is higher than `configMAX_PRIORITIES` FreeRTOS is lowering it to: `configMAX_PRIORITIES - 1` (check the file tasks.c).

This makes applications misbehave  when using multiple threads with different "cmsis-os" priorities that are higher than `configMAX_PRIORITIES`.

regards
Haithem.

rahmanih, 2018-03-28:
Hi,

did anybody look at this issue?

regards
Haithem.

ReinhardKeil, 2018-03-28:
We have looked at it already, but a solution to the problem will be provided after the Easter break.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

rahmanih, 2018-03-28:
> We have looked at it already, but a solution to the problem will be provided after the Easter break.
O.K.! thanks.

regards
Haithem.

rahmanih, 2018-04-03:
Hi Jonatan,

Can't this issue be considered as a **bug** rather than **enhancement**? 
As aforementioned, when creating cmsis_os threads with different priorities but **greater** than the `configMAX_PRIORITIES`, FreeRTOS will create tasks _**with same**_ priorities which impacts the application behaviour.

regards
Haithem.

VladimirUmek, 2018-04-04:
Hi rahmanih,

to overcome this limitation of FreeRTOS, RTOS2 API wrapper (cmsis_os2) is not using port optimised task selection. There is a little performance drawback, but on the other hand, there is no need to limit the RTOS2 API.
Please note, that FreeRTOSConfig.h configuration files that come with CMSIS-FreeRTOS RTOS2 examples contain:
```
#define configUSE_PORT_OPTIMISED_TASK_SELECTION 0
#define configMAX_PRIORITIES                    56
```
Thus all user configurable priorities in CMSIS RTOS2 can be used, from osPriorityLow to osPriorityRealtime7.

rahmanih, 2018-04-04:
great!
closing this as answered.

thanks
Haithem.

closed 2018-04-04 by rahmanih

joeytian, 2022-07-20:
Since there is an array's size determined by configMAX_PRIORITIES in FreeRTOS task.c. 

`PRIVILEGED_DATA static List_t pxReadyTasksLists[ configMAX_PRIORITIES ]`

Using CMSIS RTOS 2 with configMAX_PRIORITIES fixed to 56 will increase RAM usage.  Every priority takes ~20 bytes. This is not good for small RAM micro controllers such as some 4KB Cortex-M0 chips.