ARM-software/CMSIS-FreeRTOS #12: osEventFlagsSet fails with called from ISR

shaper-street, 2018-06-12:
if (xEventGroupSetBitsFromISR (hEventGroup, (EventBits_t)flags, &yield)  != pdFAIL) {
      rflags = (uint32_t)osErrorResource;
    } else {
      rflags = flags;
      portYIELD_FROM_ISR (yield);
    }

JonatanAntoni, 2018-06-13:
Hi @shaper-street,
thanks for reporting this issue and the fix.

It took me a really close look to spot the little difference. Hence I'd like to ask you to be slightly more verbose.

I see the problem as following:
The current implementation returns always `osErrorResource` when the actual call to `xEventGroupSetBitsFromISR` succeeds. The return value check is reversed.

VladimirUmek, 2018-08-21:
Closing issue as resolved.

closed 2018-08-21 by VladimirUmek