ARM-software/CMSIS-FreeRTOS #14: osThreadGetStackSize() implmentation missing

rahmanih, 2018-08-09:
Hi,
according to the CMSIS-RTOS v2 [documentation](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html#gab9f8bd715d671c6ee27644867bc1bf65), there the following API:
`
uint32_t osThreadGetStackSize  (  osThreadId_t thread_id )
`
this is not implemented in the FreeRTOS port, which may lead to link Errors:

<code> Error[Li005]: no definition for "osThreadGetStackSize"  </code>

regards
Haithem.

VladimirUmek, 2018-08-21:
Hi,

the documentation, under Technical data and limitation chapter, mentions limitation:
- osThreadGetStackSize is not implemented.
(See [limitations](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/dd7793adcbea0c3c0f3524f86b031ab88b9e2193/DoxyGen/General/src/cmsis_freertos.txt#L299))

Since FreeRTOS does not offer any equivalent or similar functions able to retrieve thread stack size, this function is therefore not implemented. Dummy implementation is not present in order to trigger link error you are seeing - link error gives clear indication that functionality is not present (dummy implementation might be misleading).

rahmanih, 2018-08-27:
got it thanks!

closed 2018-08-27 by rahmanih