ARM-software/CMSIS-FreeRTOS #22: HardFault Handler on prvStartFirstTask()

juliobc, 2019-03-01:
Hi,

I am creating a new test project to use FreeRtos in uvision and STM32F4 with STMCubeMX. When the basic project runs, there is a hardfault handler when it runs the prvStartFirstTask() function on the line svc 0 (Call SVC to start the first task)
I can not find the problem. Any ideas?
I attach the basic project.

[test_2.zip](https://github.com/ARM-software/CMSIS-FreeRTOS/files/2919043/test_2.zip)

Thanks

supernobby, 2019-03-04:
The problem is, that the FreeRTOS arrives at the SVC instruction in "entered critical section" state. In that, FreeRTOS disables interrupts by setting the BASEPRI register. You see this in the Register Window -> System -> BASEPRI, where the value is different from 0x00.

When the preempt priority of the SVC exception is not greater then the BASEPRI, the hardfault occurs, because the SVC handler can't execute.

A way to avoid this issue is: leave the SVC unconfigured in the STM32Cube NVIC settings, so that is has the default preempt priority of 0 at runtime.

Hope this helps.

juliobc, 2019-03-06:
Great! Solved

I set the SVC preempt priority to 14 because in RTX5 is 14. Could you to explain the difference between CMSIS-FreeRtos and CMSIS-RTX5 about SVC exception?
Should the Systick and PendSV exceptions have preempt priority to 0 too?
Thank you

supernobby, 2019-03-06:
The usage requirements for the Keil RTX5 (with CMSIS RTOS2 API) are described e. g. here: https://www.keil.com/pack/doc/CMSIS/RTOS2/html/cre_rtx_proj.html#cre_UsingIRQs

I am not so familiar with FreeRTOS. In a quick search, I did not find, that they say something about SVC preempt priority requirements.

But for the SysTick and PendSV, also FreeRTOS requires them to have the lowest priority.

But still I would say, the basic issue is, that the FreeRTOS arrives at this point in "entered critical section" state with interrupts disabled. Maybe you can discuss this with the FreeRTOS authors. If this would not be the case, the SVC preempt prio would actually not matter, as the interrupts would not be disabled like that.

juliobc, 2019-03-06:
ok, thanks

closed 2019-03-06 by juliobc