ARM-software/CMSIS-FreeRTOS #28: privileged

swordligit, 2019-03-28:


swordligit, 2019-03-28:
The MPU port define some special segment, such as privileged_functions, privileged_data, etc. is there example or demo how to define these sections?

swordligit, 2019-03-29:
Have checked the "FreeRTOS/Demo/CORTEX_MPU_Simulator_Keil_GCC/GCC_Specific", there is a ld script that define the sections privileged_functions and privileged_data.
As for privileged_data, who will copy the data from flash to RAM?

JonatanAntoni, 2019-03-29:
Hi @swordligit,

As far as I am aware of there are no provisions for loading data from flash to RAM in the GCC libraries. We have this "scatter loading" as part of the device startup components in CMSIS, e.g. in

https://github.com/ARM-software/CMSIS_5/blob/db1dc63b57abf8b7719064e5c034615c97212179/Device/ARM/ARMCM3/Source/GCC/startup_ARMCM3.c#L165-L183 

Cheers,
Jonatan

swordligit, 2019-03-30:
Thanks a lot! I will check it.

closed 2019-03-30 by swordligit