ARM-software/CMSIS-FreeRTOS #29: My cmsis-freertos porting exception randomly

swordligit, 2019-04-09:
I have ported the CMSIS-freertos  to my original cmsis-rtx software, and tested with the default example blinky.c correctly. but it is crashing randomly in my real software. 
Here is some information:
```
### EXCEPTION ###
PC =45DCD2EE, ExceptionNumber=-12
R0 =200A0064, R1 =00000000, R2 =00000000, R3 =2002D3F8
R4 =62E5DFFE, R5 =E7B24843, R6 =05A1FE4F, R7 =1DF6A8E1
R8 =0C406BE6, R9 =766302B2, R10=2000ADBF, R11=FFFED520
R12=2D3D7465, SP =200A9FE0, LR =45DCD2EF
MSP=200A9EAC, PSP=200A0064

PRIMASK=00, FAULTMASK=00, BASEPRI=00, CONTROL=00
XPSR=6100000E, APSR=nZCvq, EPSR=01000000, IPSR=0E

ICSR =04414004, AIRCR=FA050000, SCR  =00000000, CCR  =00000210
SHCSR=00070401, CFSR =00000001, HFSR =00000000, AFSR =00000000
MMFAR=E000ED34, BFAR =E000ED38
FaultInfo : (MemFault)
FaultCause: (Instruction access violation)

200A9FE0: 00000000 00F00000 00000000 2000B474
200A9FF0: 00000018 0C077DB9 0C0779A0 41000000
200AA000: 78407802 2002EA40 BF004770 BA006800
200AA010: BF004770 78007842 2002EA40 BF004770
200AA020: 47706800 70010A0B 47707043 0C0AB410
200AA030: 0A0B0E0C 700170C4 70437082 4B04F85D
200AA040: BF004770 70410A0B 47707003 0C0AB410
200AA050: 0A0B0E0C 70C17004 70837042 4B04F85D

Possible Backtrace:
 C077DB4
total task number 7 current 7

total active task 7

task#1          R       32      1393    6       S/2001e8d0      H/2001e870
app_thread      R       40      314     5       S/20020700      H/200206a4
main            R       24      265     1       S/200a9100      H/2002537c
2               X       12997   0       939708438       S/32c5  H/0
bt_main         R       32      925     7       S/20013998      H/2001393c
IDLE            R       0       491     2       S/20023ec0      H/200246c0
                B       24      381     4       S/200253f0      H/20025bf8
Tmr Svc         B       40      399     3       S/20024720      H/20024f20
task#1          S       32      1393    6       S/2001e8d0      H/2001e870
app_thread      B       40      314     5       S/20020700      H/200206a4
main            S       24      265     1       S/200a9100      H/2002537c
bt_main         S       32      925     7       S/20013998      H/2001393c
```

The thread information was printed with vTasklist, and the line 
```
2                     X       12997   0       939708438       S/32c5  H/0
```
is definitely wrong, S/xxxx is add by me with the stack base info
and H/xxxx is the thread handle.
One point is even the device crashes randomly, This line is always same!!

Another problem is there are only 7 tasks, but the vTasklist--> uxTaskGetSystemState
find 12 tasks, some task(such as main) in the different list at same time.
Do you have any advice？ thanks a lot.

swordligit, 2019-04-10:
This problem has been resolved. 
The reason is I use the wrong value of
```
configMAX_SYSCALL_INTERRUPT_PRIORITY
```
the default value is 16, this maybe come from the STM cortex-m4 chips.
my chip has 8 level interrupts and the top 3 bit of BASEPRI is valid, so
the value 16 is ignored when setting to BASEPRI, which messed the
taskENTER_CRITICAL().
Change the value to 64 resolve the problem.
Thanks anyway.

closed 2019-04-10 by swordligit

JonatanAntoni, 2019-04-10:
Hi @swordligit,

Thanks for pointing this resolution out. This `configMAX_SYSCALL_INTERRUPT_PRIORITY` option is a known failure cause when running FreeRTOS.

Can you please check if this pit fall is properly documented, somehow? Otherwise we should think about enhancing the documentation, accordingly.

Cheers,
Jonatan

swordligit, 2019-04-10:
IMO, we should check the width of register BASEPRI, becase the
configMAX_SYSCALL_INTERRUPT_PRIORITY will be set to this
register directly.
so 
```
        __set_BASEPRI(portMAX_8_BIT_VALUE);
        valid_bits = __get_BASEPRI();
```
the valid_bits can get the real valid bits of BASEPRI(for my chip, 
it is 0xe0, and the top 3 bits is valid), and then according to this, 
we can check if configMAX_SYSCALL_INTERRUPT_PRIORITY
is defined correctly, maybe assert in the early stage.