ARM-software/CMSIS-FreeRTOS #31: about CMSIS-FreeRTOS

Cherish-Gww, 2019-05-15:
I am not found memory pool management source in this project, not implemented?
I am using CMSIS OS version v1.02, the v1.02 cmsis_os.c I can find  Memory Pool Management Functions, but there is a TODO: 

//TODO
//This is a primitive and inefficient wrapper around the existing FreeRTOS memory management.
//A better implementation will have to modify heap_x.c!

My question is : 
1. is there not implement memory pool on this project, or newest CMSIS-Freertos version.
2. above TODO, this there have a patch about how to modify heap_x.c to make memory pool better?

VladimirUmek, 2019-05-15:
Hi,

no, memory pool function are not yet implemented, please see also #7 and #26.

Can you please point me to the file containing this TODO statements? Is this some modified version for the file that comes with CMSIS-FreeRTOS pack?

Cherish-Gww, 2019-05-15:
@VladimirUmek 
will CMSIS-Freertos add memory pool management recent? or never?
I got cmsis_os.c and cmsis_os.h, in stm32cubef4xx.zip, ST package a cmsis os v1.02 inside.

VladimirUmek, 2019-05-17:
I see, packages that come with the STM32CubeMX are modified by STMicroelectronics and memory pool implementation you are referring to is not part of this project.

CMSIS-FreeRTOS will add memory pool management implementation but I can't give you exact time frame as it is not in the short term plan. Depending on number of requests this might change.

Cherish-Gww, 2019-05-17:
@VladimirUmek 
Got, Thank you very much.

closed 2019-05-24 by VladimirUmek

VladimirUmek, 2019-09-24:
Commit 5db75d170849a11a4371c292b590d22e0240a44e addresses this issue.