ARM-software/CMSIS-FreeRTOS #35: NVIC_SetPriority() build error.

rahmanih, 2019-06-24:
Hi,
this [commit ](https://github.com/ARM-software/CMSIS-FreeRTOS/commit/ee8667f0178e809207b02a692899acd81d665a5f) introduced the function `SVC_Setup()` that is calling the   `NVIC_SetPriority (SVCall_IRQn, 0U);`, IIUC, the latter is device/dependent. Thus compiling the code within the STM32 Cube FW, the following errors are thrown:
``
Error[Pe223]: function "NVIC_SetPriority" declared implicitly Middlewares\Third_Party\FreeRTOS\Source\CMSIS_RTOS_V2\cmsis_os2.c 173 
``
``
Error[Pe020]: identifier "SVCall_IRQn" is undefined Middlewares\Third_Party\FreeRTOS\Source\CMSIS_RTOS_V2\cmsis_os2.c 173 
``

regards
Haithem.

VladimirUmek, 2019-06-26:
Hi,

I need to review this as the problem is more complex as it looks like.
Generally, CMSIS device header includes all core specific API and ideally we would include it into FreeRTOSConfig.h. The problem is that we can't do it and the reason is __NVIC_PRIO_BITS definition used in configMAX_SYSCALL_INTERRUPT_PRIORITY that brings unwanted 'U' literal into function xPortPendSVHandler (see port.c ) for some ports that later on leads to compilation errors on various compilers.
We can't include it into cmsis_os2.c as this brings another non-portable specifics into implementation. So
currently I am thinking about adding an include into cmsis_os2.c that would contain all portable items and thus generalize the RTOS2 implementation in cmsis_os2.c.

VladimirUmek, 2019-12-11:
So, a common (and easy) solution is to include device header into FreeRTOSConfig.h but this include must be guarded in order to not pass it to the assembler (see also [default config file](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/develop/CMSIS/RTOS2/FreeRTOS/Config/ARMCM/FreeRTOSConfig.h) for Cortex-M cores) :
`
#if (defined(__ARMCC_VERSION) || defined(__GNUC__) || defined(__ICCARM__))

#include CMSIS_device_header

#endif
`
Hopefully this resolves the problem?

rahmanih, 2020-01-20:
it does!

thanks!

closed 2020-01-20 by rahmanih

FibonacciSpiral, 2022-09-28:
What the hell is CMSIS_device_header? do you mean that literally? Please be more clear.

rkeil62, 2022-09-29:
CMSIS_device_header is a define, explained here https://arm-software.github.io/CMSIS_5/Core/html/using_pg.html