ARM-software/CMSIS-FreeRTOS #36: Outdated licensing information - please switch to Apache 2.0

pvyleta, 2019-10-27:
The Readme.md mentions that "CMSIS-FreeRTOS is licensed under the FreeRTOS license." which is however misleading information, as the original FreeRTOS is now under MIT license (since version 10.0.0), and the CMSIS part is under Apache 2.0.

As far as I understand, it would make sense to label this whole project as "licensed under Apache 2.0" as MIT is compatible with it as far as I Understand.

See https://law.stackexchange.com/questions/6081/can-i-bundle-mit-licensed-components-in-a-apache-2-0-licensed-project for discussion on that topic.

VladimirUmek, 2019-12-04:
Yes, FreeRTOS kernel is licensed under MIT and this was now corrected, see commit 71eb5b3d9594e7384d31adbf4d9582765726a6c0.

VladimirUmek, 2019-12-11:
Hi,
is there any other concern regarding licensing or can we close this issue?

closed 2020-01-06 by VladimirUmek

pvyleta, 2020-01-13:
Hello @VladimirUmek, sorry for not replying in time, thanks for updating the license info, I was considering that maybe this whole project could be labeled as "licensed under Apache 2.0" but I believe it is fine as it currently is. Thanks for the change!