ARM-software/CMSIS-FreeRTOS #38: SysTick_Handler conflict M33

raidoz, 2019-12-09:
Both cmsis_os2.c and the port.c for ARM_CM33 define SysTick_Handlers and they conflict. Older port.c's defined xPortSysTickHandler which could be chosen through config and I think cmsis_os2.c did not define a handler at all - why should it actually ... shouldn't that be left to the OS itself? The handler in cmsis_os2.c can be avoided by getting rid of the SysTick define, but that one is also deeply embedded in the core_*.h files.

What is the correct solution for using cmsis_os2.c on an M33?

VladimirUmek, 2019-12-11:
Hi,

using xPortSysTickHandler was actually more portable way of handling FreeRTOS tick which changed for Cortex M33 and M23 ports. The reason that cmsis_os2.c defines [SysTick_Handler](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/54e00ad7f1c13e73003028d0fcc1bc898341a096/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L146) is the handling of the SysTick peripheral overflow flag which needs to be cleared in order to implement [osKernelGetSysTimerCount](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/54e00ad7f1c13e73003028d0fcc1bc898341a096/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L390) function correctly (Cortex-M ports that are part of FreeRTOS do not handle this).

Using cmsis_os2.c on M33 should work out of the box. Please take a look at the [TrustZone example](https://github.com/ARM-software/CMSIS-FreeRTOS/tree/develop/CMSIS/RTOS2/FreeRTOS/Examples/TrustZone) in the CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Examples folder if you need help with the project configuration.
TrustZone example is divided into two projects and thus separating secure and non-secure side configuration.

To overcome the situation when both handlers are defined (one by M33 port and one by cmsis_os2.c), FreeRTOSConfig.h for non-secure side contains definition:

`#define SysTick_Handler                         xPortSysTickHandler`

Following the include path of the FreeRTOS config file we can see that Systick handler in port.c gets renamed and is called from cmsis_os2.c when tick interrupt is executed.
So adding this define to your config file should do the trick!

raidoz, 2019-12-11:
Thanks, the reversed define, compared to previous ports, was not immediately obvious, but the flow now makes sense. Perhaps consider documenting this somewhere, though.

closed 2019-12-11 by raidoz

VladimirUmek, 2019-12-11:
I agree, having this documented would be good, we will add this.

oblonka, 2023-02-15:
very nice!
#define SysTick_Handler                         xPortSysTickHandler