ARM-software/CMSIS-FreeRTOS #40: os_tick.h missing

doxxx, 2020-01-20:
The `os_tick.h` file is missing. There are multiple references to it in the code, causing compilation failure.

doxxx, 2020-01-20:
Nevermind, I see now that I have to include CMSIS 5 in the include path.

closed 2020-01-20 by doxxx

JonatanAntoni, 2020-01-20:
Hi @doxxx,

I see you already figured out.
May I ask you which development environment you are using?

Such issues recurrently occur if using plain software component archives, i.e. having a zip archive and adding files and include paths manually to your project. To simplify this situation we created the CMSIS-Pack standard. A standardized software pack ships some machine readable meta information with it. Hence a CMSIS-Pack compliant environment can automatically deduce which source files and include paths needs to be considered in your project.

Cheers,
Jonatan

doxxx, 2020-01-20:
I'm not using any of the vendor-provided development environments. I'm attempting to create a CMake-based project, with Visual Studio Code as my editor, so yes I am using zip archives and manually adding files and include paths.

Things were going well until I got the stage where I wanted to integrate CMSIS-RTOS2/FreeRTOS. :)

JonatanAntoni, 2020-01-21:
Okay, understood. You are giving the perfect rational why we created CMSIS-Packs ;)
Unfortunately we don't have an integration with CMake ready, yet.

doxxx, 2020-01-22:
FYI, I was able to get it working. An official CMake integration would be wonderful.

One of the issues I had was that there wasn't much explanation of `RTE_Components.h` that I could find.