ARM-software/CMSIS-FreeRTOS #41: osMutexNew() failed with 'osMutexBobust' atrribute

diverger, 2020-05-22:
When use 'osMutexNew()' with 'osMuteRobus', it will return 'NULL'. Try the code below, and you'll repeat the failure:

```
  static const osMutexAttr_t at = {NULL, osMutexPrioInherit | osMutexRecursive | osMutexRobust, NULL, NULL};
    osMutexId_t id;

    id = osMutexNew(&at);

    if (id == NULL)
    {
        printf("Failed to create a mutex\n");
    }
```

The similar code is used in Lwip's 'mem.c'.

VladimirUmek, 2020-05-22:
Hi,

unfortunately robust mutex is not supported by FreeRTOS, i.e. there is no mechanism in FreeRTOS to automatically release acquired mutex if its owner thread is terminated.
Since there is no workaround possible, osMutexNew fails and returns NULL.

This is also documented as limitation, see [Technical data and limitations](https://arm-software.github.io/CMSIS-FreeRTOS/General/html/tech_data.html#td_limitations).

Could you please provide a source of lwIP code you are referring to? I am not able to find it, but would like to see the implementation.

VladimirUmek, 2020-05-22:
Sorry, I have just found the corresponding code in lwIP.
I believe that using robust mutex in not necessary in case of lwIP, unless some reason exists that cannot be seen immediately. I would report a bug in sys_arch.c implementation.

diverger, 2020-05-22:
Yes, the code just exist in ARM's pack: "Packs\lwIP\lwIP\2.1.1\ports\cmsis-rtos\sys_arch.c".

diverger, 2020-05-22:
> Hi,
>
> unfortunately robust mutex is not supported by FreeRTOS, i.e. there is no mechanism in FreeRTOS to automatically release acquired mutex if its owner thread is terminated.
> Since there is no [...]

Are their any *elegant* way to handle this? For example, inform the user when the code compiled, instead of trhow a run-time error?

VladimirUmek, 2020-05-22:
There is no way to check if certain attribute is used within the source code, at least not from the preprocessor/compiler perspective. Other limitations related to attributes have the same problem.

Another way (besides global documentation) to slightly improve the situation would be to use function documentation (currently not present in cmsis_os2.c)  and make the limitations more obvious and visible from the source code.

diverger, 2020-05-22:
@VladimirUmek Yes, state the limitations in each function's reference will be a good idea.

diverger, 2020-05-22:
It seems the same problem exist in ARM's LwIP Pack 2.1.1 (LwIP version 2.1.2).

VladimirUmek, 2020-06-22:
Hi,
please see referenced commit and close the issue if you think that it reasonably addresses this issue.

From the information I currently have lwIP pack will be corrected in the next release. You can manually tweak current implementation and remove osMutexRobust attribute. It should work without side effects.

VladimirUmek, 2020-12-18:
This have been fixed, closing issue.

closed 2020-12-18 by VladimirUmek