ARM-software/CMSIS-FreeRTOS #43: cmsis_os2.c no os_tick.h can be find

watershade, 2020-06-03:
There is a include file "os_tick.h", but we can not find it in this package.

VladimirUmek, 2020-06-03:
OS Tick API included by os_tick.h is part of CMSIS pack. You can find the file on [CMSIS repository](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/Include/os_tick.h) for preview or download it together with the [latest CMSIS pack here](https://github.com/ARM-software/CMSIS_5/releases)

The corresponding documentation is available [using this link](https://arm-software.github.io/CMSIS_5/RTOS2/html/rtos_os_tick_api.html).

watershade, 2020-06-03:
> OS Tick API included by os_tick.h is part of CMSIS pack. You can find the file on [CMSIS repository](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/Include/os_tick.h) for preview [...]

Thanks, It is Great!

closed 2020-06-03 by watershade