ARM-software/CMSIS-FreeRTOS #44: osMutexAcquire() called before the kernel is started

aautotte, 2020-12-16:
With RTX, osMutexAcquire() returns an error when called before the kernel has been started. 
With FreeRTOS, no error is returned. 
All CMSIS_RTOS implementation should be consistent.

VladimirUmek, 2020-12-18:
Hi,
thanks for raising this issue!

According to [RTOS2 API specification](https://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__MutexMgmt.html#gabc54686ea0fc281823b1763422d2a924) behavior is not defined. We will take a look into this.

VladimirUmek, 2021-01-21:
After review I noticed that by specification osMutexAcquire is not allowed to be called before the kernel is started. Please see [documentation for osKernelStart](https://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__KernelCtrl.html#ga9ae2cc00f0d89d7b6a307bba942b5221) where it is stated that "only the functions osKernelGetInfo, osKernelGetState, and object creation functions (osXxxNew) may be called" before osKernelStart is called.

What happens if you do it anyway? The behavior is undefined but I would at least ensure that the system does not trigger any fault (HardFault etc.). Since calling osMutexAcquire before osKernelStart does not trigger the fault, no action will be taken.

aautotte, 2021-01-21:
So it is a bug in FreeRTOS implementation. If it is not allowed then the CMSIS layer should return an error.  The implementation should be consistent across the different kernels.  I can start with a FreeRTOS and then switch to another kernel and my design will be incompatible.

VladimirUmek, 2021-01-22:
FreeRTOS implementation is a bit relaxed as it does not always handle mutex owner to check access. It is not necessary wrong, we just have to be aware of behavior.
You can start with FreeRTOS and then switch to another kernel and according to CMSIS-RTOS2 API you are still not allowed to call most of the API functions before the kernel is started. So why would you do it then?

If you think that implementations shall return an error (instead of having undefined behavior) when such call is made I would suggest that you open an issue in [CMSIS 5 repository](https://github.com/ARM-software/CMSIS_5). That way your suggestion will reach broader audience that can participate in discussion.

aautotte, 2021-01-22:
>according to CMSIS-RTOS2 API you are still not allowed to call most of the API functions before the kernel is started. So why would you do it then?
This means that the CMSIS_RTOS2 implementation for FreeRTOS is wrong. It should return an error when the kernel is not started. Like it happens with CMSIS-RTOS2 implementation for Micrium and RTX.  The CMSIS-RTOS2 implementation for FreeRTOS cannot be more relaxed than the other implementations. This kind of limitation is not well defined in the API as you realized yourself so if I start my testing my design using FreeRTOS and not realize such limitation then when I switch to another kernel It doesn't work.

RobertRostohar, 2021-01-25:
As explained before, CMSIS-RTOS2 specifies that only certain functions may be called before osKernelStart. It does not specify what happens if functions which are not allowed to be called are still called but leaves this to the implementation. Therefore FreeRTOS implementation is compliant.

closed 2021-05-17 by VladimirUmek