ARM-software/CMSIS-FreeRTOS #45: osMessageQueueGet(timeout = osWaitForever) always failed with return code = osErrorParameter since IS_IRQ() predicates Truere

Yajun-Hu, 2021-02-03:
1. I'm using CMSIS-FreeRTOS-10.3.1 with a Cortex-M33-NTZ implementation;
2. simple application: 2 tasks, one message sender, another message receiver, same task priorities;
3. the message receiver would stuck in its loop without yielding
4. here is pseudo code of its loop:
` void task_receiver(void *parameter)
{
    while (true) {
      if (osOK == osMessageQueueGet(mq_id, &msg, 0, osWaitForever)) {
        // do stuff
      }
    }
}`

thanks for your support!

Yajun-Hu, 2021-02-03:
by using FreeRTOS api replacing the cmsis_os2 one, this issue is gone.
```C/C++
void task_receiver(void *parameter)
{
    while (true) {
      if (pdTRUE == xQueueReceive(mq_id, &msg, osWaitForever)) {
        // do stuff
      }
    }
}
```

but why?

VladimirUmek, 2021-02-05:
Hi,

for Cortex-M33 IS_IRQ() would translate into

`
#define IS_IRQ_MASKED()           ((__get_PRIMASK() != 0U) || (__get_BASEPRI() != 0U))
`
`
#define IS_IRQ_MODE()              (__get_IPSR() != 0U)
`
`
#define IS_IRQ()                          (IS_IRQ_MODE() || (IS_IRQ_MASKED() && (KernelState == osKernelRunning)))
`

meaning that you are either in interrupt or your interrupts are disabled when the kernel is running. 

Looking at the [core register values](https://developer.arm.com/documentation/dui0552/a/the-cortex-m3-processor/programmers-model/core-registers?lang=en) might reveal what is going on.

Yajun-Hu, 2021-02-07:
VladimirUmek ,

first of all, thanks for prompt response.
I still have 2 doubts:
1. if in interrupt, we should call xReceiveQueueFromISR() instead of xReceiveQueue(), others API calling would be failed, but now xReceiveQueue() calling is okay;
2. if we disable interrupts at application level, we still can call RTOS APIs, right? 

best.

VladimirUmek, 2021-02-09:
1. xReceiveQueue does not perform any check if the function is called from interrupt or not. Its implementation is different and is meant to be only called from thread while xReceiveQueueFromISR() is designed to work properly when called from interrupt. xReceiveQueue might still return pdTRUE when called from interrupt but this is not safe way to do it.

2. correct, although we must pay attention to which function is allowed to be called from interrupt

VladimirUmek, 2021-05-17:
Closing as no further input is provided.

closed 2021-05-17 by VladimirUmek