ARM-software/CMSIS-FreeRTOS #46: osThreadNew creates a high-priority task, which is scheduled first and then returns to osThreadId_t

zbianbiaos, 2021-03-19:
In such a situation, there is already one byte of data in uart rx fifo. The low-priority task sys_up_task calls osThreadNew to create the high-priority task uart_rx_task, used to starts the uart interrupt , turns on uart rx and dispatch rx data. At this time, osThreadNew function does not return uart_rx_task's handle, but the interrupt happened and sends this byte to uart_rx_task. Because the handle is not obtained, this byte of data is lost

VladimirUmek, 2021-05-17:
Hi,
can you give more context? How design threads that handle uart operations relate to CMSIS-FreeRTOS implementation?

zbianbiaos, 2021-05-20:
```c
#include <cmsis_os2.h>

static osThreadId_t tid;

static void uart_rx_cb(uint8_t *data, uint16_t len)
{
    // for example set flag to thread 'tid'
    osThreadFlagsSet(tid, 1);
}

static void console_func(void *arg)
{
    printf("tid=%p\n", tid);  // now the tid is NULL until cpu schedule to system_init func to finish osThreadNew and return tid
    uart_init();
    uart_rx_register(uart_rx_cb);
    uart_open_read();

    for (;;)
    {
        osThreadFlagsWait(1, osFlagsWaitAny, osWaitForever);

        /*
         * handle data
         */
    }
}

void init_console(void)
{
    osThreadAttr_t tattr =
    {
        .name = "console",
        .priority = 9,
        .stack_size = 4096,
    };
    // create console task priority 9
    tid = osThreadNew(console_func, NULL, &tattr);
}

void init_system(void *arg)
{
    /*
     * init others ...
     */

    init_console();

    /*
     * init others ...
     */

    osThreadExit();
}

void main(void)
{
    osThreadAttr_t tattr =
    {
        .name = "system",
        .priority = 5,
        .stack_size = 4096,
    };

    // create init_system task priority 5
    osThreadNew(init_system, NULL, &tattr);
}

```

As shown above, an **init_system** task is created has a priority of 5 in func **main**

**init_system** calls **init_console** to create **console_func** task, priority 9.

When the console_func has not run to the osThreadFlagsWait, tid is still NULL

At this time tid = osThreadNew(console_func, NULL, &tattr) this code has not been executed yet, only when the system is scheduled back to init_system, osThreadNew will return tid after execution is complete

Is it possible to add ThreadId_t to attr to solve this?

VladimirUmek, 2021-05-20:
I guess osThreadGetId would solve your issue:

```c
static void console_func(void *arg) {
  osThreadId_t tid = osThreadGetId();

  printf("tid=%p\n", tid);
  ...
}
```
or alternatively
```c
static void console_func(void *arg) {

  printf("tid=%p\n", osThreadGetId());
  ...
}
```

zbianbiaos, 2021-05-24:
OK,thanks a lot

closed 2021-05-24 by zbianbiaos