ARM-software/CMSIS-FreeRTOS #47: FreeRTOS API functions must not be called from within a critical section.

zbianbiaos, 2021-03-19:
In the cmsis_os2.c code, IS_IRQ_MASKED is used to determine whether context  in interrupt or critical. Returning true will execute the *fromISR function, but in the FreeRTOS description, [FreeRTOS API functions must not be called from within a critical section](https://freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html)
. Use IS_IRQ_MASKED to execute fromISR in the critical section, which will cause inexplicable problems. I hope to use configASSERT to prohibit this operation.thanks

VladimirUmek, 2021-05-17:
Hi,
I'm not sure if I understand the issue. Would you like to use configASSERT to catch the situation when RTOS2 function is called from the critical section?
Can you please provide further explanation or give an example?

zbianbiaos, 2021-06-07:
Fine, I will take more care about it, thanks

closed 2021-06-07 by zbianbiaos