ARM-software/CMSIS-FreeRTOS #49: vQueueAddToRegistry  should not be called if name== NULL

giso-c, 2021-04-16:
In following code

   #if (configQUEUE_REGISTRY_SIZE > 0)
      if (hMutex != NULL) {
        if (attr != NULL) {
          name = attr->name;
        } else {
          name = NULL;
        }
        vQueueAddToRegistry (hMutex, name);
      }
      #endif

vQueueAddToRegistry  is called even when name == NULL but it should not as in vQueueAddToRegistry , name == NULL is used to define a free slot, In latest code of freertos an assert has been added
    void vQueueAddToRegistry( QueueHandle_t xQueue,
                              const char * pcQueueName ) /*lint !e971 Unqualified char types are allowed for strings and single characters only. */
    {
        UBaseType_t ux;

        configASSERT( xQueue );
        configASSERT( pcQueueName );

We have same issue for other call to vQueueAddToRegistry

paulbartell, 2021-04-19:
I rolled back the change to the FreeRTOS Kernel to maintain backwards compatibility in commits [46f7feba815b49fca5f5884304a465844812f1ef](https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/46f7feba815b49fca5f5884304a465844812f1ef) and [71f5af4e0f8d5fad2c4d83c43aa6748eb6cfaaf9](https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/71f5af4e0f8d5fad2c4d83c43aa6748eb6cfaaf9). 

Calling vQueueAddToRegistry with a NULL pcQueueName parameter will no longer result in an assertion, but will also no longer result in an invalid item added to the Queue Registry.

VladimirUmek, 2021-05-17:
Hi,
thank you for reporting this issue. The CMSIS RTOS2 wrapper has been modified and will now no longer call vQueueAddToRegistry when name argument is NULL. Please check referenced commit.

closed 2021-06-02 by VladimirUmek