ARM-software/CMSIS-FreeRTOS #50: Incorrect timeout handling in osThreadFlagsWait with osFlagsWaitAll

rtoijala, 2021-06-21:
The timeout handling in `osThreadFlagsWait` when using `osFlagsWaitAll` is incorrect and can lead to dramatically reduced timeouts.

Imagine a task waiting on two flags A and B with a timeout of 1000ms. The flag A is never set. The flag B is set once every 1ms.  `osThreadFlagsWait` calls `xTaskNotifyWait (0, clear, &nval, tout);`, where `tout` is initially 1000ms. `xTaskNotifyWait` wakes up after 1ms when B is set. `tout` is then updated as follows:

```c
td = xTaskGetTickCount() - t0;
if (td > tout) {
    tout  = 0;
} else {
    tout -= td;
}
```
`t0` is the starting time of `osThreadFlagsWait`, so `tout` becomes 999ms.

`xTaskNotifyWait` is called again in a loop, so the next time B is set (2ms into the 1000ms timeout), `tout` becomes 997ms (instead of the correct 998ms). The third time (3ms into the 1000ms timeout), it becomes 997ms - 3ms = 994ms.

In other words, during each iteration, the remaining timeout `tout` is reduced by the total time waited, not by the time waited in that iteration. In the example above this results in the 1000ms timeout occurring after only 46ms have actually passed.

One possible fix is to change `if (td > tout)` to `if (td > timeout)` and `tout -= td` to `tout = timeout - td`.

VladimirUmek, 2021-07-07:
Many thanks for reporting this issue!

The fix, as suggested, was added in the referenced commit. Please take a look and close the issue if you think that the issue is resolved.

rtoijala, 2021-07-07:
The fix looks good to me. Thank you!

closed 2021-07-07 by rtoijala