ARM-software/CMSIS-FreeRTOS #53: how can I avoid the crash?

rameble, 2021-12-07:
now I run a testcase which cause freertos reboot: 

the testcase source code is as below：  

1     osStatus_t uwRet;
 2    g_cmsisMutexId = osMutexNew(NULL);
 3    uwRet = osMutexDelete(g_cmsisMutexId);
 4    uwRet = osMutexDelete(g_cmsisMutexId);

when running at 4th line, system will reboot。 

Could you please tell me how can I know g_cmsisMutexId is invalid after executiong 3rd line?  thanks a lot.

rameble, 2021-12-08:
thanks for your quick reponse. 

The testcase belongs to thirdparty test suite. I can not modify the testcase.  So I want to konw whether I can check g_cmsisMutexId is invalid after executiong 3rd line.  So I can modify the implementation of the osMutexDelete.

ReinhardKeil, 2021-12-08:
To my knowledge FreeRTOS as no provisions to protect from such scenarios. IMHO it would require changes to the RTOS Kernel itself.   An alternative implementation is RTX - and it should prevent from this type of problems.

rameble, 2021-12-08:
@ReinhardKeil 
thanks a lot.  what does RTX mean?

JonatanAntoni, 2021-12-08:
Hi @rameble,

RTX5 is the reference implementation for the CMSIS-RTOS2 API, you can find it as part of CMSIS5.
The documentation is at https://arm-software.github.io/CMSIS_5/develop/RTOS2/html/rtx5_impl.html.

As Reinhard pointed out the RTX5 implementation has double-free protection built-in. FreeRTOS seems not.
Where does this test code come from? It looks somewhat invalid to me as it expects double-free protection which I think is not guaranteed by the API.

Be aware of another limitation/difference: The RTX5 implementation implicitly unblocks all waiting threads upon mutex deletion. FreeRTOS *explicitly* states one **must not** delete a mutex/semaphore "that has tasks blocked on it"! The CMSIS-RTOS2 API is *not* specific in this situation.

Cheers,
Jonatan

rameble, 2021-12-09:
Hi @JonatanAntoni @ReinhardKeil 
It seems that RTX5 is the implementation standard of CMSIS-RTOS2.   It it not concrete implementation.  Does this opensource project have plan or already to support RTX5? 

does CMSIS standard Committee provide official test suite? otherwise I have not evidence to  say to thirdparty the testcase  is invalid. 

Thanks a lot.

JonatanAntoni, 2021-12-09:
Hi @rameble,

The basic idea behind CMSIS-RTOS2 API is to have the actual implementation to be replaceable. You could try to replace CMSIS-FreeRTOS with RTX5. This should work as long as you didn't access any FreeRTOS features directly.

Regarding an official test suite we are planning to release one, but this has not been done, yet.

I don't expect any official testsuite could give you evidence that the testcase is invalid. The latest official documentation for [osMutexDelete](https://arm-software.github.io/CMSIS_5/5.8.0/RTOS2/html/group__CMSIS__RTOS__MutexMgmt.html#gabee73ad227ba4587d3db12ef9bd582bc) does not make any assumptions regarding double-free. But it clearely states

> After this call [note: osMutexDelete], the mutex_id is no longer valid and cannot be used. 

This could be translated to: "Calling any API function with an invalid mutex_id is undefined behaviour."
Hence, the testcase relys on "undefined behaviour" which I'd consider "invalid".

Cheers,
Jonatan

rameble, 2021-12-09:
Thanks a lot. Your suggestion is very helpful.

 I have no idea on the  workload  of  replacing FreeRTOS with RTX5.   Are there any workload on hardware apdation or  kernel adaption?

JonatanAntoni, 2021-12-09:
HI @rameble,

The workload depends mainly on two aspects:
- Development environment you are using. In Keil MDK replacing CMSIS-FreeRTOS with RTX5 is just a view clicks.
- Coupling of the application with details of the RTOS implementation. This can range from nearly no additional effort to lots of effort.

Sorry for not being more specific. I'd need to know much more details about your project.

Cheers,
Jonatan

rameble, 2021-12-09:
Hi @JonatanAntoni ,
thanks for your patient reponse. 

Cheer. 
Ramble

closed 2022-03-03 by VladimirUmek