ARM-software/CMSIS-FreeRTOS #60: Auto clear overflow flag in SysTick_Handler(cmsis_os2) will break `configUSE_TICKLESS_IDLE 1` in freeRTOS

yyjdelete, 2022-11-19:
Seems SysTick_Handler in cmsis_os2 will auto clear overflow flag for SysTick_Handler
https://github.com/ARM-software/CMSIS-FreeRTOS/blob/9cf2e4b4f621abfed0ee2de421d31988f8b1b0f6/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L152-L158

But it will also break the default implementation of `vPortSuppressTicksAndSleep` (`#define configUSE_TICKLESS_IDLE 1`), which required to check the same flag, and it make the wakeup timer almost frozen.
https://github.com/ARM-software/CMSIS-FreeRTOS/blob/3da36094e21d6fed61719ecf9bb3530a11fd2bb0/Source/portable/GCC/ARM_CM3/port.c#L552-L566

VladimirUmek, 2022-11-22:
Hi, thanks for bringing that up!
SysTick_Handler in cmsis_os2 clears overflow flag to enable correct functionality of osKernelGetSysTimerCount. It does not seem like there will be an easy workaround. We need to investigate this issue and see what can be done.

VladimirUmek, 2022-11-24:
Hi,
please see the referenced commit:
SysTick_Handler was updated:
https://github.com/ARM-software/CMSIS-FreeRTOS/blob/13acfbef7be85119fc6bc56832c455d4547d92c7/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L156-L161

and as a consequence also osKernelGetSysTimerCount had to be updated:

https://github.com/ARM-software/CMSIS-FreeRTOS/blob/13acfbef7be85119fc6bc56832c455d4547d92c7/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L475-L516

Do you have any concerns related to the implemented workaround?

closed 2022-12-01 by VladimirUmek