ARM-software/CMSIS-FreeRTOS #65: 10.5.1 regression when the firmware is loaded from a bootloader

escherstair, 2023-01-13:
I have an application that can be compiled in two different ways:

- to be executed directly (from the Cortex-M reset)
- to be started from a bootloader (i.e., vector table relocation, ...)

Everything works fine with CMSIS-FreeRTOS 10.4.6, but when I upgraded the pack to 10.5.1 I see a regression:

- when the application is executed directly, everything is ok
- when it's started by a bootloader, this is the call chain
`main() >> HAL_Init(); >> HAL_InitTick(TICK_INT_PRIORITY)`
where
`#define  TICK_INT_PRIORITY            (15UL) /*!< tick interrupt priority */`
Then
```
 /* Configure the SysTick IRQ priority */
  if (TickPriority < (1UL << __NVIC_PRIO_BITS))
  {
    HAL_NVIC_SetPriority(SysTick_IRQn, TickPriority, 0U);
    uwTickPrio = TickPriority;
  }
```
and after that the function `void TIM6_DAC_IRQHandler(void)` is continously called and nothing else.

I suppose some issue with interrupt priorities.
Does 10.5.1 require a different priority configuration based on 10.4.6?
Is there anything else I can do to debug this regressioin?

VladimirUmek, 2023-01-18:
Hi,
issue #55 was the change that could have the effect on interrupts. I don't think anything else was changed that could have side effects.

escherstair, 2023-01-18:
It's what I think too, but I can't find which is the change that could have impacted.

VladimirUmek, 2023-02-15:
Did you find the root cause for this problem? Was the problem solved?
Can you provide more info, otherwise I'll have to close this issue due to the inactivity.

escherstair, 2023-02-15:
As a short term workaround I had to downgrade to CMSIS-FreeRTOS 10.4.6, because I have to keep the pace of development.
But for sure I'm going to investigate this in the next few weeks (I hope at the beginning of March), because I need to find the root cause.
Keep open this issue (if possible).
Thanks

VladimirUmek, 2023-02-15:
Okay, if possible it would be great to know what causes the difference in behavior - will keep the issue open.

CHAMSTM, 2023-08-14:
Hi, Is it possible please to have some more details about the issue ? 
- Which STM32 is used ?
- Which Bootloader is used ? (STM32 system bootloader? or custom bootloader running from user flash ?)

Were you able to identify if the problem was related to NVIC priority group configuration ? or to specific Interrupt (ie. TIM6) ?

VladimirUmek, 2023-11-06:
Closed due to inactivity.

closed 2023-11-06 by VladimirUmek

escherstair, 2023-11-06:
Hello @VladimirUmek and @CHAMSTM 
sorry for my late feedback.
I have some more deatils, and I'm not sure if this helps or not:

- I use STM32H725, with peripheral configuration code generated by STM32CubeMX 6.9.2. I think that STM32CubeMX is important here (I will describe the reason later)
- I use a custom bootloader, running from user flash

I confirm that the call chain is
`main() >> HAL_Init(); >> HAL_InitTick(TICK_INT_PRIORITY)`
and then something unexpected happens.

STM32CubeMX 6.9.2 generates source code to use `TIM6` as a time base source.

I saw that inside `stm32h7xx_hal.c`
`__weak HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority)`
is defined as a `__weak` function
and it's reimplemented by source code generated by STM32CubeMX 6.9.2 in `stm32h7xx_hal_timebase_tim.c`

In `stm32h7xx_hal.c` I see
```C
/* Configure the SysTick IRQ priority */
if (TickPriority < (1UL << __NVIC_PRIO_BITS))
{
  HAL_NVIC_SetPriority(SysTick_IRQn, TickPriority, 0U);
  uwTickPrio = TickPriority;
}
else
{
  return HAL_ERROR;
}
```

In `stm32h7xx_hal_timebase_tim.c` I see
```C
/*Configure the TIM6 IRQ priority */
if (TickPriority < (1UL << __NVIC_PRIO_BITS))
{
  HAL_NVIC_SetPriority(TIM6_DAC_IRQn, TickPriority ,0U);
  /* Enable the TIM6 global Interrupt */
  HAL_NVIC_EnableIRQ(TIM6_DAC_IRQn);
  uwTickPrio = TickPriority;
}
else
{
  return HAL_ERROR;
}

/* Enable TIM6 clock */
__HAL_RCC_TIM6_CLK_ENABLE();
```

You can see the difference `SysTick` vs `TIM6`, but this is not an issue itself.
Most important, see that in `stm32h7xx_hal_timebase_tim.c` there is
```C
/* Enable the TIM6 global Interrupt */
HAL_NVIC_EnableIRQ(TIM6_DAC_IRQn);
```
As sson as this is called, the callback `void TIM6_DAC_IRQHandler(void)` is called, and I think this is because the clock has been already enabled by the bootloader, and/or a TIM6 interrupt is pending.

Not sure why everything work with CMSIS-FreeRTOS 10.4.6, but the code generated by STM32CubeMX doesn't convince me.
I'll go on with my investigation.