ARM-software/CMSIS-FreeRTOS #66: Do you have ESP-IDF adaptation

mantriyogesh, 2023-01-27:
I tried to port using your lib with FreeRTOS using ESP-IDF.
Getting below error:

```
/Users/yogesh/code/h1/esp_hosted_fg/host/components/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c:26:10: fatal error: cmsis_compiler.h: No such file or directory
 #include "cmsis_compiler.h"             // Compiler agnostic definitions
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

1. Is this mandatory file to port?
2. If yes, do you have any [ESP-IDF](https://github.com/espressif/esp-idf/tree/release/v5.0) adaption for this?

VladimirUmek, 2023-02-15:
Hi,
cmsis_compiler.h is part of [CMSIS ](https://github.com/ARM-software/CMSIS_5) core and is used to handle CPU specific intrinsics for different compilers. When one is using CMSIS everything should work out of the box but I don't think ESP-IDF is using CMSIS and there is no adaptation available...

mantriyogesh, 2023-02-15:
Sure, thanks.

closed 2023-02-15 by mantriyogesh