ARM-software/CMSIS-FreeRTOS #77: xQueuePeek not implemented

pedroazeredo04, 2023-08-03:
Hi! I am using a simples FreeRTOS CMSIS-V2 application for a robotracer with STM32, and i missed the implementation of a `xQueuePeek` function in the CMSIS-V2 adaptation.

Is there any equivalent of this function in the CMSIS-V2 adaptation?

JonatanAntoni, 2023-08-07:
Hi @pedroazeredo04,

An equivalent to `xQueuePeek` is not available in CMSIS-RTOSv2 API.
The queue api in CMSIS assumes a unique producer-consumer relation. In use cases with multiple consumers "receiving" a message from the queue without removing it would break this uniqueness.

May I ask you what you try to achieve? I.e., why would you need to know about the next message in the queue *before* actually receiving it?

-Jonatan

pedroazeredo04, 2023-08-09:
Thank you for answering @JonatanAntoni,

As I said, the system I'm trying to build is a simple Robotracer software. In this software, i publish the information of the sensors in a unitary queue, for instance, the line sensors publish in the `lineSensorQueue`, which holds only one message. Other tasks that use that information may pull the message from the queue. However, when two different tasks try to pull the same message, one of them get an error, because the queue is already empty. So our team tried to use the Peek function instead of the Get, as it would solve the problem, but we found out there was no equivalent in the CMSIS Wrapper.

I think the solution we'll come across will be to implement our own `osQueuePeek` in our code. Thank you again for answering

JonatanAntoni, 2023-08-09:
Hi @pedroazeredo04,

Okay, if I read this correctly, you use the message queue not as a real queue but like a shared buffer? That means you remove the single message once a new value becomes available? And all other tasks are only peek'ing into the queue to read the current sensor value?

If this is the case, I don't see why you are using a queue. If this is really only a single value at a time, you could simply use shared storage protected by a mutex. This might be less overhead.

Of course, you can always bypass RTOS2-APIs and use native API (FreeRTOS in your case). But you'll loose the RTOS abstraction. For a specific application like yours, which is not very likely to be ported to another RTOS, this might be acceptable. Once it comes to reusable software components which shall not depend on a specific RTOS implementation (which was the driver beging CMSIS-RTOS APIs) we'd need to look for another solution.

Perhaps @VladimirUmek will add a comment whether adding a queue peek function to API would be feasible.

-Jonatan