ARM-software/CMSIS-FreeRTOS #87: [Updates]: support new FreeRTOS 11.0.x & CMSIS RTOS2 2.3.0

rahmanih, 2024-02-29:
Hi @VladimirUmek,
Is there any plan to update the CMSIS-FREERTOS against the new [CMSIS RTOS2 (2.3.0)](https://arm-software.github.io/CMSIS_6/latest/RTOS2/index.html)  and FreeRTOS-Kernel [V11.0.x](https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V11.0.0)

thzanks & regards
Haithem.

VladimirUmek, 2024-03-04:
Hi Haithem,

update to FreeRTOS-Kernel V11.0.x is currently in progress and will hopefully be available soon. Initially it was planed to synchronize the CMSIS-FreeRTOS pack with CMSIS 6.0.0 release but it way delayed due to unforeseeable events.

At this point the RTOS2 wrapper implementation will not implement [Process Isolation functions](https://arm-software.github.io/CMSIS_6/latest/RTOS2/CMSIS_RTOS_ProcessIsolation.html). This should not represent a problem as most applications will continue to use RTOS2 API without Process Isolation.

VladimirUmek, 2024-04-15:
CMSIS-FreeRTOS 11.0.1 has now been finally published, hence I'm closing this issue. Please open new issues for potential problems.
Thank you for your patience!

closed 2024-04-15 by VladimirUmek