ARM-software/CMSIS-FreeRTOS #96: osPriorityISR priority can be configured by user

CezaryGapinski, 2024-03-28:
`osThreadNew()` and `osThreadSetPriority()` functions allow to set `osPriorityISR`:

https://github.com/ARM-software/CMSIS-FreeRTOS/blob/main/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L545C1-L548C8

https://github.com/ARM-software/CMSIS-FreeRTOS/blob/main/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L720C1-L722C4

Is it correct?

`configMAX_PRIORITIES` must be set to 56, but priorities in `osPriority_t` enum are numbered from 0. For current implementation `osPriorityISR = 56` might set index outside the priorities array.

VladimirUmek, 2024-04-10:
Hi,

you are right, thank you for reporting this.
Priorities equal or higher than configMAX_PRIORITIES are limited to configMAX_PRIORITIES-1 in FreeRTOS, but mapping to RTOS2 priority defines in not correct.

I'll check again and provide a fix.

VladimirUmek, 2024-04-10:
The mapping of priorities is now changed, FreeRTOS priority now equals (osPriority_t - 1).
Since this only shifts priority values side effects are not expected.

closed 2024-04-10 by VladimirUmek