ARM-software/CMSIS-FreeRTOS #109: Implementation of osKernelRestoreLock is dangerous and might leave the kernel locked

pakluba, 2024-07-22:
https://github.com/ARM-software/CMSIS-FreeRTOS/blob/ba82e7e1f62e72313992e47b88b442b40131bbdb/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L397

Unlike RTX, FreeRTOS does nesting in vTaskSuspendAll(). If the scheduling has already been suspended before osKernelLock() is called, osKernelRestoreLock() will increase the nesting level by 1 instead of removing 1 lock. So:

```
vTaskSuspendAll();      // the nesting level is at least 1
...
state = osKernelLock(); // does not increase nesting, returns 1 (was locked already)
...
osKernelRestoreLock();  // will call vTaskSuspendAll() to "restore" the lock -> boom...
```

VladimirUmek, 2024-07-22:
Hello,
good find, thanks! I'll review the implementation and provide a fix.

VladimirUmek, 2024-08-19:
Thanks you for going forward and providing the PR! It has been merged hence I'm closing this issue.

closed 2024-08-19 by VladimirUmek