ARM-software/CMSIS-FreeRTOS #110: Support portable for ARM CA7

SteMMo33, 2024-07-29:
Hi all,
what about a portable branch toward ARM CA7 - i'm working with a STM32MP135.

Thanks!

VladimirUmek, 2024-07-30:
Hi,
FreeRTOS currently does not have a portable for Arm Cortex-A7 core - see [the FreeRTOS-Kernel repository](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable). You could probably use and adopt the port for Cortex-A9 which exists.
If you want to trigger the request please check the FreeRTOS-Kernel repository [and open an issue there](https://github.com/FreeRTOS/FreeRTOS-Kernel/issues). Thanks!

SteMMo33, 2024-07-31:
I added the Cortex-A9 files, but I have the following errors with the assembler:

```
Building file: ../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S
Invoking: GNU Arm Cross Assembler
arm-none-eabi-gcc -mcpu=cortex-a7 -mthumb -O2 -g -x assembler -MMD -MP -MF"src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.d" -MT"src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.o" -c -o "src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.o" "../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S"
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S: Assembler messages:
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:77: Error: selected processor does not support `fmrxne R1,FPSCR' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:147:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:79: Error: selected processor does not support `vpushne {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:147:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:80: Error: selected processor does not support `vpushne {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:147:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:109: Error: selected processor does not support `vpopne {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:157:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:110: Error: selected processor does not support `vpopne {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:157:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:112: Error: selected processor does not support `vmsrne FPSCR,R0' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:157:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:109: Error: selected processor does not support `vpopne {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:167:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:110: Error: selected processor does not support `vpopne {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:167:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:112: Error: selected processor does not support `vmsrne FPSCR,R0' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:167:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:77: Error: selected processor does not support `fmrxne R1,FPSCR' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:261:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:79: Error: selected processor does not support `vpushne {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:261:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:80: Error: selected processor does not support `vpushne {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:261:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:109: Error: selected processor does not support `vpopne {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:278:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:110: Error: selected processor does not support `vpopne {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:278:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:112: Error: selected processor does not support `vmsrne FPSCR,R0' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:278:  Info: macro invoked from here
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:304: Error: selected processor does not support `fmrx R1,FPSCR' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:305: Error: selected processor does not support `vpush {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:306: Error: selected processor does not support `vpush {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:313: Error: selected processor does not support `vpop {D16-D31}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:314: Error: selected processor does not support `vpop {D0-D15}' in ARM mode
../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.S:315: Error: selected processor does not support `vmsr FPSCR,R0' in ARM mode
make: *** [src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/subdir.mk:34: src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/portASM.o] Error 1
```
I tested both arm and thumb mode but same errors.
I'm wondering if they are strictly CA9 core commands ..

JonatanAntoni, 2024-07-31:
Hi @SteMMo33,

The instructions are related to FPU on Cortex-A. Looks like some settings don't match in your case. Try to add `-mfpu=neon-vfpv4` flag to enable NEON instructions via VFPv4 FPU.

SteMMo33, 2024-07-31:
@JonatanAntoni thanks!
I added -mfloat-abi=hard -mfpu=neon-vfpv4 options.

Now my problem is the error:

```
Building file: ../src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.c
Invoking: GNU Arm Cross C Compiler
arm-none-eabi-gcc -mcpu=cortex-a7 -mthumb -mfloat-abi=hard -mfpu=neon-vfpv4 -O2 -g -DconfigUSE_TASK_FPU_SUPPORT=2 -I"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS2\Drivers\CMSIS\RTOS2\Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\Source\include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\RTOS2\Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\Source\portable\Gcc\ARM_CA7" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\Device\ST\STM32MP13xx\Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\RTOS2\FreeRTOS\Config" -std=gnu11 -MMD -MP -MF"src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.d" -MT"src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.o" -c -o "src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.o" "../src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.c"
../src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.c:41:7: error: expected ';' before 'void'
   41 | __WEAK void software_init_hook (void) {
      |       ^~~~~
      |       ;
make: *** [src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/subdir.mk:29: src/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Source/clib_os.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
```

Maybe the _WEAK symbol is wrong?

JonatanAntoni, 2024-07-31:
@VladimirUmek may comment on where `__WEAK` is expected to be defined.

VladimirUmek, 2024-08-01:
WEAK symbol comes from CMSIS device header included [by FreeRTOSConfig.h](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/main/CMSIS/RTOS2/FreeRTOS/Config/FreeRTOSConfig.h). CMSIS_device_header is defined in RTE_Components.h (generated by CMSIS build tool).

JonatanAntoni, 2024-08-01:
The compiler command line dumped above does not contain the usual `_RTE_` defines. Hence, I suspect the device header is not included at all.

VladimirUmek, 2025-04-16:
Closing issue - in general requests for new ports should go to FreeRTOS kernel repository.

closed 2025-04-16 by VladimirUmek