ARM-software/CMSIS-FreeRTOS #111: Missing os_ticks.h

SteMMo33, 2024-07-30:
Hi all.
I'm trying to compile CMSIS-FreeRTOS inside my simple GCC project for STM32MP135.
When I compile a file I have:

```
Building file: ../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.c
Invoking: GNU Arm Cross C Compiler
arm-none-eabi-gcc -mcpu=cortex-a7 -mthumb -O2 -g -I"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS2\Drivers\CMSIS\RTOS2\Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\Source\include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\Device\ST\STM32MP13xx\Include" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS" -isystem"C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\RTOS2\FreeRTOS\Config" -std=gnu11 -MMD -MP -MF"src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.d" -MT"src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.o" -c -o "src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.o" "../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.c"
In file included from C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\Source\include/FreeRTOS.h:64,
                 from ../src/CMSIS-FreeRTOS/Source/portable/Gcc/ARM_CA7/port.c:34:
C:\Progetti\dbTechnologies\testSTM32\testSTM32_CMSIS_RTOS_FreeRTOS\src\CMSIS-FreeRTOS\CMSIS\RTOS2\FreeRTOS\Config/FreeRTOSConfig.h:354:12: fatal error: os_tick.h: No such file or directory
  354 |   #include "os_tick.h"
      |            ^~~~~~~~~~~
compilation terminated.
```

where I can find this file?
Thanks

JonatanAntoni, 2024-07-30:
The `os_tick.h` interface header is from
https://github.com/ARM-software/CMSIS_6/blob/main/CMSIS/RTOS2/Include/os_tick.h

SteMMo33, 2024-07-30:
Thanks a lot!

closed 2024-07-30 by SteMMo33