ARM-software/CMSIS-FreeRTOS #113: What tags to use for FreeRTOS 10.6.X and 11.0.1

PhilippFinke1988, 2024-09-20:
Hello there,

I'm using this repository along with the official [FreeRTOS kernel repository](https://github.com/FreeRTOS/FreeRTOS-Kernel) to build FreeRTOS + CMSIS RTOS2 API as a [Conan](https://conan.io/) package. My package supports all FreeRTOS versions starting from 10.6.0. For this purpose, I try to use tags from this repository that matches the corresponding FreeRTOS kernel, but it seems there are no tags for 10.6.X and 11.0.0.

What should I use for these kernel versions?

VladimirUmek, 2024-09-23:
Hello,

unfortunately there is no CMSIS-FreeRTOS release that would be based on FreeRTOS kernel 10.6.x, hence there is no tags. There were updates to the code but those versions were not released. We only tag versions that are validated before the release therefore any intermediate versions may not be of production quality.

VladimirUmek, 2024-09-23:
The difference between FreeRTOS kernel 11.0.0 and 11.0.1 is the SBOM file. That won't break anything.

PhilippFinke1988, 2024-09-23:
Perfect, thanks you. That information is all I need.

closed 2024-09-23 by PhilippFinke1988