ARM-software/CMSIS-FreeRTOS #114: WIP- CMSIS

araujorayza, 2024-10-23:
I am using NUCLEO-F756ZG with freeRTOS to communicate using ethernet. 

Using the CubeIDE debug feature, I figure out that the program is halting at
[image]

JonatanAntoni, 2024-10-28:
Hi @araujorayza,

That means, in your scenario `pxQueue->uxItemSize` is different from zero?
We'd need to figure out how that can happen. 

Can you provide the full project? Or even better, a minimal example to reproduce the issue?

Thanks,
Jonatan

VladimirUmek, 2025-04-16:
Closing issue - the issue seems to be part of larger application where xQueueSemaphoreTake is called with invalid object.

closed 2025-04-16 by VladimirUmek