ARM-software/CMSIS-FreeRTOS #115: osCMSIS and osCMSIS_FreeRTOS are missing in V11.x

EmergReanimator, 2024-11-18:
osCMSIS and osCMSIS_KERNEL (osCMSIS_FreeRTOS) are part of CMSIS-RTOS specification:
https://arm-software.github.io/CMSIS_5/RTOS/html/group__CMSIS__RTOS__KernelCtrl.html#gab78dce646fabec479c5f34bc5175b7de

It would be great add `CMSIS-FreeRTOS/tree/v11.1.0/CMSIS/RTOS2/FreeRTOS/Include/cmsis_os2.h` header with these defines.

JonatanAntoni, 2024-11-18:
@EmergReanimator, please take note that by design in CMSIS-Packs one must not duplicate an API header into multiple locations. The API is specified at a unique location and shipped in the CMSIS base pack. All implementations of that API are expected to refer to the unique location of the API header.

EmergReanimator, 2024-11-18:
> @EmergReanimator, please take note that by design in CMSIS-Packs one must not duplicate an API header into multiple locations. The API is specified at a unique location and shipped in the CMSIS base [...]

That's totally fine. But the `soCMSIS` and `osCMSIS_FreeRTOS` are gone since `CMSIS-RTOS V1` support was dropped in V11.x;

Previously these defines were located here: https://github.com/ARM-software/CMSIS-FreeRTOS/blob/v10.5.1/CMSIS/RTOS2/FreeRTOS/Include1/cmsis_os.h

VladimirUmek, 2024-11-19:
Hi,
defines osCMSIS and osCMSIS_FreeRTOS (osCMSIS_KERNEL) are part of CMSIS-RTOS1 implementation and support for RTOS1 specification was dropped, hence these defines were also removed.

CMSIS-RTOS2 specification does not have such "identification" defines. It does have [osKernelGetInfo ](https://arm-software.github.io/CMSIS_6/latest/RTOS2/group__CMSIS__RTOS__KernelCtrl.html#ga6f7764e7250c5c5364c00c45a5d1d199) which only provides information at run time.

I'm not sure why exactly were you using these defines, but maybe the below defines that reside in FreeRTOS kernel (task.h) could help:
```
#define tskKERNEL_VERSION_NUMBER       "V11.1.0"
#define tskKERNEL_VERSION_MAJOR        11
#define tskKERNEL_VERSION_MINOR        1
#define tskKERNEL_VERSION_BUILD        0
```
If you maintain an old project that still uses RTOS1 API, you can still use CMSIS-FreeRTOS 10.5.1. I don't think that latest FreeRTOS kernel packed into CMSIS-FreeRTOS 11.0.1 and higher contains functionality that would have any meaningful performance/stability impact.

closed 2024-11-19 by EmergReanimator