ARM-software/CMSIS-FreeRTOS #127: CM33 MPU without TrustZone

stejcon, 2025-02-26:
I am trying to use a CM33 chip that has MPU support but does not have the TrustZone extension. It seems that the PDSC file only supports building v8m MPU support if TrustZone is also enabled. From what I know this isn't a requirement of FreeRTOS. I am trying to build with GCC. When trying to build, I get linker errors from cmsis_os2.o and clib_os.o that a number of functions are undefined e.g. MPU_xTaskGetCurrentTaskHandle, MPU_xTaskGetSchedulerState

Line 1151 has `<component Cgroup="Core" Cvariant="Cortex-M Non-Secure MPU" condition="FreeRTOS Core TZ MPU">`, but there doesn't seem to be a corresponding "FreeRTOS Core MPU" condition to use the files from Source/portable/GCC/ARM_CM33_NTZ instead of Source/portable/GCC/ARM_CM33

Line 913 has `<component Cgroup="Core" Cvariant="Cortex-M" isDefaultVariant="true" condition="FreeRTOS Core CM">` which does use the NTZ files but it doesnt include the mpu wrappers file. Nothing uses Source/portable/GCC/ARM_CM33_NTZ/non_secure/mpu_wrappers_v2_asm.c.

Is there a correct way in a cproject file to enable MPU support with the mpu wrappers without having TrustZone? Adding the mpu wrappers file to the Cortex-M Cvarient works and successfully builds, but I can't imagine that this something someone hasn't tried to do before.

VladimirUmek, 2025-03-10:
Hi, 
thanks for pointing this issue. Apparently with the integration of reworked MPU support in FreeRTOS v8M cores were not added to the pdsc. This will be fixed - most likely v8M cores will be added to the ["Cortex-M MPU" variant](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/6a7cfa9ee58adb282e267d579204aa9234f3eff7/ARM.CMSIS-FreeRTOS.pdsc#L1005), which is basically the same as Cortex-M variant with MPU wrappers added.

In the mean time, your workaround is fine.

VladimirUmek, 2025-04-16:
Cortex-M MPU variant now supports v8M cores without TrustZone.

closed 2025-04-16 by VladimirUmek